johnboiles/esp-roomba-mqtt #1: Suggestions for improvement of Readme

Landrash, 2017-09-12:
Suggest the following be added to the Readme.
 - Picture of setup.
 - [Schematic](http://fritzing.org/home/) of wiring.
 - Link to platforms where this can be used.

nunofgs, 2017-10-03:
I would love do this for my roomba. Can you upload wiring instructions?

bachulator, 2017-11-07:
It would be great to have those covered.

nunofgs, 2017-11-07:
I would also love a blog post or two on how you reverse engineered the roomba protocol.

Share the knowledge! :)

Adminius, 2017-11-08:
> I would also love a blog post or two on how you reverse engineered the roomba protocol.

http://www.robotappstore.com/files/KB/Roomba/Roomba_SCI_Spec_Manual.pdf

brubaked, 2018-07-18:
Even a labeled picture of your setup would be very helpful for those of us who are just starting out.

johnboiles, 2018-10-30:
@kg333 was kind enough to do the schematic!