johnboiles/esp-roomba-mqtt #2: Failed to read sensor values from Roomba

bachulator, 2017-12-05:
Platform: Roomba 780
board: d1_mini
Used PlatformIO to build the code.
Start/stop/return_to_home seems to work just fine but unfortunately no telemetry. Connecting back to telnet debug port getting repeating:
(d) Failed to read sensor values from Roomba

Happy to provide any necessary information

Iture, 2017-12-21:
I have the same problem - Roomba 786p, NodeMCU. Commands work, telemetry doesn't. Looks like serial read times out.

andenmad, 2018-02-09:
hi 

Anyone found a solution for this i expiriens same problem

Daigaard, 2018-03-13:
We have found that if you use an ESP8266 module (Eg: WEMOS D1 Mini or NODE MCU), with a 470 Ohm (or equivalent) resistor mounted on the TX pin, this resistor needs to be removed in order to get the sensor values to the ESP. As this resistor pull's the TX pin to 3.3V. 

By removing this resistor you can no longer program the module trough USB.

mannkind, 2018-03-17:
For what it's worth, my Wemos D1 Mini couldn't read the Roomba sensors either. I found this snippet in an [iRobot Create 2 Doc](http://www.irobot.com/~/media/MainSite/PDFs/About/STEM/Create/Create_2_to_5V_Logic.pdf)

> In some cases, the drive strength of a Roomba TX line is not enough to drive the RX line of another board (for example, in some revisions of Arduino). In this case, a simple PNP transistor (2N2907A, 2N3906, or 2N4403, among others) can be used to provide enough “drive” for the Arduino.

I used a 2N3906 and following the circuit as described in the PDF; I can now read sensor values from the Roomba!

se4n01, 2018-07-05:
So instead of removing the 470O resistor you used the 2N3906? I tried adding it to the voltage divided RX/TX line and it did not work. Did you remove the voltage divider?

mannkind, 2018-07-05:
@se4n01 - I am not using a voltage divider, that's correct.

johnboiles, 2018-10-31:
I suspect this is a big reason the serial transmission seems unreliable.

johnboiles, 2018-10-31:
I'll try the 2N3906 thing in a couple days. Looks just as simple if not simpler to wire than the voltage divider.

se4n01, 2018-11-01:
keep us posted, the 2N3906 did not work for me sadly

johnboiles, 2018-11-01:
@se4n01 interesting thanks for the tip. Maybe the best route is to use a bidirectional level shifter and do this right. That way we can eliminate any uncertainty about voltage levels being the cause of issues. I think I have a couple of these laying around 

https://www.sparkfun.com/products/12009

I’ll have to move to a wemos d1 mini or some other board with a 5v rail but that’s not the worst thing

mannkind, 2018-11-01:
It's unclear to me from @se4no1's post and I'm not an expert in this stuff, but the 2N3906 does require a resistor (~ 1-10k) between the base pin and the mcu pin.

johnboiles, 2018-11-05:
Cross-posting here. I tried the bidirectional level shifter and it did not work. I have no idea why it didn't. I wrote in a bit more detail about it [here](https://community.home-assistant.io/t/add-wifi-to-an-older-roomba/23282/91).

se4n01, 2018-11-07:
Hi guys, thanks for sharing that HA thread, it's very informative, thank you! Shame the level shifter did not work but I think it's for the best to use it anyway. I also in the end tried removing the 470Ohm resistor, it also did not work. I will try playing around with the code this weekend when I have some time. As an aside, NiMH batteries actually like to be deep discharged, partial discharging causes them to lose the memory of the zero and will reduce capacity with time.

johnboiles, 2018-11-07:
Interesting to know about the deep discharge. I think 10.8v is a deep-ish discharge for the roomba battery right?

dinkyshaw, 2018-12-09:
Hi folks, just sharing another data point for troubleshooting the Rx sensor issue. Thanks for writing the code! It's been a fun project so far (my first microcontroller project) but the lack of sensor data has been a disappointment. 

I have it integrated with the [MQTT vacuum component](https://www.home-assistant.io/components/vacuum.mqtt/) in Home Assistant and can get it to respond to all commands over MQTT but it would be nice to have the battery level information to set up automations to send it back to dock to charge and then send it back out when it's done.

Platform: Roomba 400 series (model 4225)
board: NodeMCU LUA CP2102 ESP-12E
wiring diagram:
[https://drive.google.com/file/d/1tdn0-SuSc2DnK_XVjq6-IIlq8nxzRpOp/view?usp=sharing](url)

** I have also tried wiring with a logic level converter between the NodeMCU board and the Roomba on both Rx and Tx instead of using the PNP transistor (2N3906) and, like the PNP transistor wiring, can Tx to but not Rx from Roomba.

**platformio.ini** file:
```
[env:d1_mini]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps =
  RemoteDebug
  PubSubClient
  ArduinoJson

upload_port = roomba.local
build_flags = -DLOGGING=1 -DMQTT_MAX_PACKET_SIZE=512
```
I also had to set the baud rate to 57600 in the **main.ccp** file for the 400 series Roomba to respond to commands:
```
// Roomba setup
Roomba roomba(&Serial, Roomba::Baud57600);
```
I see no payload at all when monitoring the **'vacuum/state'** MQTT topic :(

I've also tried using the [Arduino sketch](https://github.com/thehookup/MQTT-Roomba-ESP01) written by the [The Hook Up in his Roomba setup](https://www.youtube.com/watch?v=t2NgA8qYcFI&t=381s) and, again, Roomba will respond to commands but I can't get sensor data.

Any ideas are greatly appreciated but mostly wanted to share my experience. Thanks again!

Armadill0, 2020-03-28:
@dinkyshaw Got my Roomba 780 perfectly working with this wiring: https://github.com/thehookup/MQTT-Roomba-ESP01/blob/master/schematic.JPG