johnboiles/esp-roomba-mqtt #10: TX and BRC

kriskirkland, 2018-10-30:
Device: Roomba 700 using ESP01
I have one problem and one question; 
1) RX works fine I get all the sensor values but nothing happens when sending it any commands. 

Question: Can I switch the BRC ping from GPIO15 to GPIO2 since the esp01 doesnt expose 15?

johnboiles, 2018-10-31:
Huh. To get sensor values you also need to send data over serial to tell the Roomba to send them, so that suggests to me that your TX to the Roomba works. What exactly do you mean by 'sending it any commands'? Are you posting to an mqtt topic?

You should be able to switch the BRC pin just fine. There's nothing special about that pin.

kriskirkland, 2018-10-31:
I guess what I should have said was that I can received sensor stats via the telnet debug interface but when i send a start command as referenced in the readme "mosquitto_pub -t 'vacuum/command' -h $MQTT_SERVER -p 1883 -u $MQTT_USER -P $MQTT_PASSWORD -V mqttv311 -m "turn_on"" nothing happens regardless if its on the dock or not. I have also tried to send it "return_to_base" while cleaning and it does nothing. It is publishing its stats to MQTT as well.

johnboiles, 2018-10-31:
OK so you can

    mosquitto_sub -t 'vacuum/state' -v -h $MQTT_SERVER -p 1883 -u $MQTT_USER -P $MQTT_PASSWORD -V mqttv311

And you get something like

    vacuum/state {"battery_level":100,"cleaning":false,"docked":true,"charging":true,"voltage":16894,"current":0,"charge":2214}

But you can't 

    mosquitto_pub -t 'vacuum/command' -h $MQTT_SERVER -p 1883 -u $MQTT_USER -P $MQTT_PASSWORD -m 'turn_on' -V mqttv311 -r -q 0

That is strange. Do you see in the debug messages anything about receiving the command?

kriskirkland, 2018-11-01:
I found the problem, mosquitto_pub wouldn't work with the quotes on the command topic. IE: This doesnt work  `mosquitto_pub -t 'vacuum/command' -h $MQTT_SERVER -p 1883 -u $MQTT_USER -P $MQTT_PASSWORD -V mqttv311 -m "turn_on"`

But his does work: 
`mosquitto_pub -t vacuum/command -h $MQTT_SERVER -p 1883 -u $MQTT_USER -P $MQTT_PASSWORD -V mqttv311 -m "turn_on"`

closed 2018-11-01 by kriskirkland

kriskirkland, 2018-11-01:
Client machine is Windows 10, broker is Debian 9.