johnboiles/esp-roomba-mqtt #14: Roomba wakes up and restarts every 10 seconds

kaosrq, 2019-01-06:
When connecting the Nodemcu the roomba enters a loop that wakes it up (sound included) and after about 10 seconds it restarts and returns to do the same. I can have debug connection by telnet. But when sending a command I receive the disconnect message. When trying to press the buttons of the roomba, he receives the commands but executes them with these 10-second pauses. By executing a pin to the ip of the nodemcu, I check that the board restarts at that time. Test with 2 other plates and the result is the same. Even if the feed for usb restarts every 10 seconds.

kaosrq, 2019-01-06:
Checking the serial monitor gives me this:

Exception (0):
epc1=0x4000dce5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffffc60 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffe00:  4000066d 9d438d71 000026fe 00000098  
3ffffe10:  00000041 9d438d71 00000020 fffffffe  
3ffffe20:  00000066 3fffc6fc 2c448d71 00985297  
3ffffe30:  00418937 002d1666 3ffeffd4 00000030  
3ffffe40:  0000001f 40104bf5 3ffefa6c 4020b96c  
3ffffe50:  00000000 3ffefa6c 3ffefa6c 40203c02  
3ffffe60:  4000050c 00000000 0000001f 40104bf5  
3ffffe70:  4020bdc8 3ffffe80 000000c8 00000008  
3ffffe80:  3ffffe70 00000000 00000001 00000000  
3ffffe90:  3fffdab0 00000000 3fffd9d0 3ffeffd4  
3ffffea0:  00000000 00000000 00000001 3ffe862c  
3ffffeb0:  00000000 3fffdad0 3ffeffd4 00000030  
3ffffec0:  00000000 000026e5 3ffeffd4 00000030  
3ffffed0:  00000000 00000000 00000001 3ffeffcc  
3ffffee0:  00000000 3fffdad0 3ffeffd4 00000030  
3ffffef0:  40201d55 3fffc6fc 3ffefd54 4020b808  
3fffff00:  00000000 4bc6a7f0 00002711 402064d0  
3fffff10:  00000000 00000000 4bc6a7f0 00000000  
3fffff20:  3ffefccc 00000064 4010663a 00000000  
3fffff30:  00000000 00000015 3ffefa6c 40204a1b  
3fffff40:  00000000 4bc6a7f0 00002711 3ffeffd4  
3fffff50:  00000000 00000000 4bc6a7f0 00000000  
3fffff60:  3fffdad0 3ffeffd4 4010663a 051eb851  
3fffff70:  00000000 00000064 3ffefa6c 3ffefd30  
3fffff80:  00002711 3ffefa6c 3ffefd54 40203ee4  
3fffff90:  00000000 00000000 00000000 4020b5a0  
3fffffa0:  00000000 00000000 00000001 3ffeffd4  
3fffffb0:  3fffdad0 00000000 3ffeffcc 4020987c  
3fffffc0:  feefeffe feefeffe 3ffe862c 40100721  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

ljmerza, 2019-01-10:
I get the same issue. Any help?

ljmerza, 2019-01-10:
my stack trace

`0x4020382c: loop() at C:\Users\Leo\Desktop\esp-roomba-mqtt-master\src/src.ino line 417
0x4020382c: loop() at C:\Users\Leo\Desktop\esp-roomba-mqtt-master\src/src.ino line 417
0x40208c3c: WiFiClient::operator bool() at C:\Users\Leo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WiFi\src\WiFiClient.cpp line 350
0x401001c0: millis at C:\Users\Leo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266\core_esp8266_wiring.c line 183
0x40206c78: RemoteDebug::handle() at C:\Users\Leo\Documents\Arduino\libraries\RemoteDebug\RemoteDebug.cpp line 317
0x40208d40: Roomba::pollSensors(unsigned char*, unsigned char, unsigned char*) at C:\Users\Leo\Documents\Arduino\libraries\Roomba\Roomba.cpp line 279
0x401001c0: millis at C:\Users\Leo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266\core_esp8266_wiring.c line 183
0x402083f4: esp_yield() at C:\Users\Leo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266\core_esp8266_main.cpp line 91
0x40201546: delay at C:\Users\Leo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266\core_esp8266_wiring.c line 54
0x4020391c: loop() at C:\Users\Leo\Desktop\esp-roomba-mqtt-master\src/src.ino line 451
0x40202944: mqttCallback(char*, unsigned char*, unsigned int) at C:\Users\Leo\Desktop\esp-roomba-mqtt-master\src/src.ino line 136
0x40208ac4: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\leo\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 1934
0x40208aac: std::_Function_handler ::_M_invoke(std::_Any_data const&, char*, unsigned char*, unsigned int) at c:\users\leo\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x402084a0: loop_wrapper() at C:\Users\Leo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266\core_esp8266_main.cpp line 125`

kaosrq, 2019-01-14:
The problem seems to occur once you connect to the broker, loses the connection after showing this message:

```
 Debug: Command recevied: p
* Show profiler: On
(D p:^3698ms) Reconnecting MQTT
(D p:^0000ms) Attempting MQTT connection...
(D p:^0105ms) MQTT connected
```
If wrong MQTT data is configured (preventing it from connecting) the crash does not occur.

kaosrq, 2019-01-14:
The problem seems to arise in the function `void sendStatus()` . When assigning a fixed value to the line ` root["battery_level"] = (roombaState.charge * 100)/roombaState.capacity;`  The crash does not happen. Probably at some point `roombaState.capacity` is 0 and gives a division error by 0.

Mikei386, 2019-01-19:
I have exakt the same problem.
Had to delete the line to get it to work.
Now of course Home Assistant doesn't get a battery reading.

Using telnet to see the values, there are all there and stable.
Thinking about doing the math in Home Assistant to get the %.

Mikei386, 2019-01-19:
For now I'm using this "ghetto-fix":
I looked up the value for Capacity via telnet and substituted "roombaState.capacity;"

`  root["battery_level"] = (roombaState.charge * 100)/2696; `

I know the capacity will change over time but for now, I can get an estimate till the problem is fixed...

Daigaard, 2019-02-10:
Thanks for the post, this help me a bit further. Now my ESP (WeMos D1) is no longer rebooting. 
But i only get empty values in my MQTT status. :
{"battery_level":0,"cleaning":false,"docked":false,"charging":false,"voltage":0,"current":0,"charge":0}
And it is only send once. 

When i log into the ESP using Telnet, and set VERBOSE logging i get valid data:
(V t:84011ms) Got Packet of len=17! Distance:0mm ChargingState:4 Voltage:14416mV Current:-203mA Charge:2687mAh Capacity:2696mAh

Any help on how get the data send to MQTT broker?

Steffen

johnboiles, 2019-02-10:
Sounds like we should just put an `if` check if `roombaState.capacity` is 0 so we don't divide by 0 (what @kaosrq said). Not sure if that will solve all of the problems in this thread but can't hurt.

Daigaard, 2019-02-11:
Problem Solvej: The MQTT string (when populated with values) was longer than the default setting of 128 In the MQTT library. When I changed the value to 256 I started getting MQTT statuses send to the broker.
**Next challenge:**
Battery was drained from testing, when I placed roomba in charger, it stopped responding and I could not ping it😧. This project just won’t let me move on....
Well guess it’s a Hardware problem, this I should be able to solve on my own...😀

ljmerza, 2019-05-05:
did you change `MQTT_MAX_PACKET_SIZE` to fix this?