johnboiles/esp-roomba-mqtt #25: build problem

pilot1981, 2019-07-09:
Hi,
using suggested Visual studio Code, I obtain this error during build task:

src/main.cpp:1:10: fatal error: 
      'Arduino.h' file not found
#include <Arduino.h>

I need to download libraries from other repository?

mrgluek, 2019-07-09:
did your set your board in platformio.ini? (`[env:...]`)
also i need to change `lib_deps` to:
```
lib_deps =
  RemoteDebug
  PubSubClient
  ArduinoJson@~5.13.4,!=6```

pilot1981, 2019-07-11:
yes, I tried to setup "esp01" board....

so I need to wait your change in code in GitHub?

pilot1981, 2019-07-11:
today, I tried to use arduino IDE v.1.8.9 and build command return this error:

Arduino:1.8.9 (Mac OS X), Scheda:"Generic ESP8266 Module, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), ck, 26 MHz, 40MHz, QIO (fast), 512K (no SPIFFS), 2, nonos-sdk 2.2.1 (legacy), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

main:408:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6
   StaticJsonBuffer<200> jsonBuffer;
           ^
exit status 1
StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

Questo report potrebbe essere più ricco di informazioni abilitando l'opzione
"Mostra un output dettagliato durante la compilazione"
in "File -> Impostazioni"

mrgluek, 2019-07-11:
Here's the fix:
https://github.com/johnboiles/esp-roomba-mqtt/pull/26/commits/fb7fc07b708dd024e75ee172a91aabc4c67604de