johnboiles/esp-roomba-mqtt #40: Strange Thing taking from Roomba

CsepregiArtur, 2021-03-04:
Hello

I have the following issue: from time to time my Roomba is saying something in docking position, "lei aler" maybe???
I don't have any clue what is he saying.
I changed the software on my ESP 01 and tried a simple sketch and when the Roomba is Docked and I send again the seak dock command to it it's saving the same thing."lei aler"
Why is the command to seek Dock is sent from time to time if the Roomba is in docking position?
Can I disable this? And How?

blizarazu, 2021-03-07:
What is your Roomba model? I have a 780 and it does not say anything. Maybe you can modify the code and check if it is already docked when you send the seek dock command and if so, avoid sending the serial signal.

CsepregiArtur, 2021-03-07:
Good idea

I have model 560 :)

I will modify the code thank you !

closed 2026-06-26 by CsepregiArtur