johnboiles/esp-roomba-mqtt #42: Future maintenance & ESP32?

aly-fly, 2024-12-16:
Hi, is this project still maintained? I see useful pull requests being ignored and last commits done in 2019.
I have my [own version](https://github.com/aly-fly/Roomba-ESP32-MQTT-SmartNest-WebServer) of MQTT Roomba from long ago. Now I am looking to upgrade my Roombas with a better HA support and this project looks promising. I would adjust it to run also on ESP32. I am willing to contribute here, but if this is abandoned, I will rather make my copy...

iago-veiga, 2024-12-27:
Hi! I'm not entirely sure, but as you can see from the open and closed PRs, the last contribution was in 2019.

If you decide to work on something, let me know, and I’ll try to help.

There’s a discontinued project that used ESPHome, in case it’s helpful: https://github.com/mannkind/ESPHomeRoombaComponent

aly-fly, 2024-12-29:
Thanks for the feedback. I also found that repo and a few others. EspHome is good because it automatically adds a Device and Entities to HA. For MQTT system it's either fully manual or you have to implement the HA Discovery messages. Lots of precise work in any case. 
I will pick one of the working EspHome repositories and upgrade that one.

closed 2024-12-29 by aly-fly